adafruit/Adafruit_Learning_System_Guides #2346: Missing board pins in pi_radio/rfm69_check.py

mraiser, 2022-12-02:
The button pins D5, D6, and D12 are not exposed in the "board". Work-around is to import "microcontroller" and use microcontroller.pin.D5, microcontroller.pin.D6, and microcontroller.pin.D12.

mraiser, 2022-12-02:
For reference, the board is Raspberry Pi Zero W v1.1 with the RFM69 900MHz Bonnet.

mraiser, 2022-12-02:
If useful:

> cat /proc/device-tree/model
Raspberry Pi Model B Rev 2

> history | grep pip
   33  sudo apt install python-pip
   34  sudo apt install python3-pip
   35  sudo pip3 install --upgrade adafruit-python-shell
   36  sudo pip3 install --upgrade adafruit-blinka
   37  sudo pip3 install --upgrade setuptools
   38  pip3 install adafruit-circuitpython-bus_device
   39  pip3 install adafruit-circuitpython-busdevice
   40  pip3 install adafruit-circuitpython-register
   66  pip3 install adafruit-circuitpython-ssd1306
   67  sudo pip3 install adafruit-circuitpython-framebuf
   68  sudo pip3 install adafruit-circuitpython-rfm69

makermelissa, 2024-03-25:
The pins are exposed in https://github.com/adafruit/Adafruit_Blinka/blob/main/src/adafruit_blinka/board/raspberrypi/raspi_40pin.py It's possible this was being detected incorrectly at the time (Raspberry Pi Zero W v1.1 vs Raspberry Pi Model B Rev 2). However, I updated PlatformDetect in May 2023 which decodes the Rev codes and improves the detection a bunch, which likely fixed this. We can re-open if not.

closed 2024-03-25 by makermelissa