adafruit/Adafruit_Learning_System_Guides #2370: Rotary Trinkey examples invert the button state

DeeHants, 2022-12-30:
The Rotary Trinkey Arduino surface dial, and volume examples both invert the button state.
```
  bool curr_button = !digitalRead(PIN_ENCODER_SWITCH);
  ...
  if (curr_button && !last_button) { // switch pressed!
    Serial.println("Play/Pause");
    Consumer.write(MEDIA_PLAY_PAUSE);
  }
```

This makes the software think the button is pressed in bootup, and physically pressing it, reports as releasing, then re-pressing it.
```
11:10:05.767 -> Release
11:10:05.907 -> Press
11:10:10.135 -> Release
11:10:10.320 -> Press
```

The `!` should be removed from the `curr_button` assignment line.

Affected examples:
* [`surfacedial_encoder_demo.ino`](https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/main/Rotary_Trinkey/surfacedial_encoder_demo/surfacedial_encoder_demo.ino)
* [`volumemute_encoder_demo.ino`](https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/main/Rotary_Trinkey/volumemute_encoder_demo/volumemute_encoder_demo.ino)
* The same exaples on the [sites](https://learn.adafruit.com/adafruit-rotary-trinkey/arduino-examples)

DeeHants, 2022-12-30:
It looks like both the CircuitPython examples have the same issue, but I am not in a position to fix or test those.
* [`CircuitPython_Volume_Knob_Example`](https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/main/Rotary_Trinkey/CircuitPython_Volume_Knob_Example/code.py)
* [`CircuitPython_YouTube_Frame_Example`](https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/main/Rotary_Trinkey/CircuitPython_YouTube_Frame_Example/code.py)

closed 2022-12-30 by dhalbert

dhalbert, 2022-12-30:
> It looks like both the CircuitPython examples have the same issue, but I am not in a position to fix or test those.

These do work, but the logic is a little confusing. Also the Volume control one does PLAY/PAUSE, but the comments say it mutes. I'll look into cleaning these up -- thanks.

PR #2371 (2022-12-30): Correct the rotary trinkey examples inverting the button state. Fixes…
Corrects the rotary trinkey examples inverting the button state.
Fixes #2370
commit 4afe72fe (2022-12-30): Correct the rotary trinkey examples inverting the button state. Fixes #2370