adafruit/Adafruit_Learning_System_Guides #2385: Glitch when trying to display a dash

Szabolcs2008, 2023-01-17:
Hi! 
I started working with the 7 segment display.
I downloaded the tutorial code, and tried to display a "-" symbol.

But it turned out like this:
(the other cables are for a temperature sensor and a fan)
https://imgur.com/jLiU4ut

The code:
```
# SPDX-FileCopyrightText: 2019 Mikey Sklar for Adafruit Industries
#
# SPDX-License-Identifier: MIT

import time
import board
import busio
from adafruit_ht16k33 import segments

# Create the I2C interface.
i2c = busio.I2C(board.SCL, board.SDA)

# Create the LED segment class.
# This creates a 7 segment 4 character display:
display = segments.Seg7x4(i2c)

# Clear the display.
display.fill(0)

display.print("-")
time.sleep(0.5)
```

I fixed it with set_digit_raw and it seems like it uses a different binary code than the one specified in the program.

The working code (still the same tutorial code edit):

```
# SPDX-FileCopyrightText: 2019 Mikey Sklar for Adafruit Industries
#
# SPDX-License-Identifier: MIT

import time
import board
import busio
from adafruit_ht16k33 import segments

# Create the I2C interface.
i2c = busio.I2C(board.SCL, board.SDA)
# Create the LED segment class.
# This creates a 7 segment 4 character display:
display = segments.Seg7x4(i2c)

# Clear the display.
display.fill(0)

display.set_digit_raw(3, 0b01000000)
time.sleep(0.5)
````

mikeysklar, 2023-01-17:
Thank you for pointing out the dash not printing and using the set_digit_raw() work around with the correct binary value. 

I'll wire up my hardware and probably end up submitting a PR to the appropriate github repo (ht16k33 circuitpython library) if I find the same behavior. 

Were you running this on a Raspberry Pi? Which mod of 7-segment display were you using? Was it the .56" height?

mikeysklar, 2023-01-17:
I went ahead and loaded the code on a Pi4 using a .56" 7-seg display and was able to reproduce the same issue you are seeing. Based on the way the library has been re-written it makes more sense to use the set_digit_raw() as you have done.

Szabolcs2008, 2023-01-18:
I used this guide:

https://learn.adafruit.com/matrix-7-segment-led-backpack-with-the-raspberry-pi/7-segment-test

Szabolcs2008, 2023-01-18:
I found another error:

When i try to display a float type number, the part after the colon gets cut off
(still using the code from the guide)

```
import time
import board
import busio
from adafruit_ht16k33 import segments

i2c = busio.I2C(board.SCL, board.SDA)

number = 36.45

display = segments.Seg7x4(i2c)
display.fill(0)

display.print(number)
```

result:
https://imgur.com/e00LAPo

Fix:
If i first convert it to a string with the ```str()``` function, it works fine
```
display.print(str(number))
```
result:
https://imgur.com/ci7HCg3

mikeysklar, 2023-01-26:
Thank you for your assistance with both issues. I've modified the example code sevensegment_test/code.py to not stumble over the '-' and to cast the floating point numbers to strings. 

I've submitted a PR to the github repo so the replacement code should be up in the next day.