adafruit/Adafruit_Learning_System_Guides #2581: Package the code for the AI-storybook?

anotherthomas, 2023-07-29:
Hi, do you have any plans to package the magic_ai_storybook code for pypi? That would make installation easier.

makermelissa, 2023-08-14:
I hadn't considered doing that. I'll take a look into what will be needed to create a package.

anotherthomas, 2023-08-14:
Would you accept pull requests? Although I've restructured the code quite a bit, so it wouldn't just be packaging...

makermelissa, 2023-08-14:
Sure, I can take a look at it. However, since there's a guide attached to this code that would likely need to be updated, depending on the amount of restructuring, it might be a bit before I can get around to testing and reviewing it.

anotherthomas, 2023-08-14:
No worries, just wanted to  avoid double effort. I'm on it anyway for a present :)