adafruit/Adafruit_Learning_System_Guides #2631: Sorry, we had to truncate this directory to 1,000 files. 47 entries were omitted from the list.

dglaude, 2023-09-23:
I don't know if there is an easy way to do that without breaking things.
But it would great to have less folder is the root.
Maybe having a folder for 3D printing guide, or splitting Arduino and CircuitPython or I don't know.

I am a frequent user of github.com web interface to navigate, so this is not helping.

TheKitty, 2023-09-25:
We have had some discussions on this - you might bring it up in the Monday
meeting long discussions

TheKitty, 2023-09-25:
I am a heavy web interface user myself.

Yes, if you could post more as to how things are difficult or discuss in CircuitPython Discord long discussion please. 

It's a known issue.

dglaude, 2023-09-25:
https://github.com/adafruit/adafruit-circuitpython-weekly-meeting/blob/main/2023/2023-09-25.md

TheKitty, 2023-11-15:
This has been started. Slowly, subdirectories are being made for major Adafruit products. The README is keeping tally of the directories moved. It's not a fast process as the corresponding guides must be hand edited. We should get below 1,000 soon.

closed 2023-12-04 by ladyada