adafruit/Adafruit_Learning_System_Guides #2654: PiCowbell Adalogger CircuitPython example need updating for PCF8523

caternuson, 2023-10-27:
Re this thread:
https://forums.adafruit.com/viewtopic.php?t=205469

These examples:
https://github.com/adafruit/Adafruit_Learning_System_Guides/tree/main/PiCowbell_Adalogger_Examples

Looks like the layout for the CircuitPython PCF8523 library changed, so the import syntax needs updating.

nhedler-cubeworks, 2025-02-06:
It looks like this specific example was fixed here: https://github.com/adafruit/Adafruit_Learning_System_Guides/commit/4d3513ace602af732d9a53012fa43617212cf0ed

and this should get the rest of them: https://github.com/adafruit/Adafruit_Learning_System_Guides/pull/2960

caternuson, 2025-02-07:
On neat! Thanks for the additional fixes! Should have considered looking at other examples when I opened this originally. 

Going to close this since the original issue is resolved.

closed 2025-02-07 by caternuson