adafruit/Adafruit_Learning_System_Guides #2715: The link to lindustrial pack by oberza is broken

ggsato, 2024-01-21:
The link in the following warning of [Next Level MakeCode Arcade](https://learn.adafruit.com/next-level-makecode-arcade-games/asset-tool) is broken.

> For this background art, I adapted some of the excellent designs of OllieBerzs' Industrial Pack assets found here https://ollieberzs.itch.io/industrial-pack You can name your own price and use them in personal and commercial projects.

The correct link seems to be this one.
https://oberzs.itch.io/industrial-pack

I will make a PR.

ggsato, 2024-01-21:
I can't find the guide [Next Level MakeCode Arcade Games](https://learn.adafruit.com/next-level-makecode-arcade-games) in this repository. Doesn't this belong to this one?

TheKitty, 2024-01-21:
You can report broken links via the Feedback  link in any guide. https://learn.adafruit.com/feedback/new?page_id=16161

ggsato, 2024-01-21:
@TheKitty Thanks, done!

closed 2024-01-21 by ggsato

jedgarpark, 2024-01-24:
I updated the guide with the new link, thanks @ggsato !

ggsato, 2024-01-24:
@jedgarpark Confirmed. Thanks for your awesome guides!

jedgarpark, 2024-01-25:
@ggsato I appreciate it, you're very welcome!