adafruit/Adafruit_Learning_System_Guides #2742: WalkMP3rson Suggested Changes

mikeysklar, 2024-02-28:
Forum user [@galinawind provided a detailed word document of suggested improvements and some corrections](https://forums.adafruit.com/viewtopic.php?t=208693) for the WalkMP3rson guide.  

The document is attached here, but I let an AI summarize the contents.

[Walkmp3rson_project_corrections.docx](https://github.com/adafruit/Adafruit_Learning_System_Guides/files/14439276/Walkmp3rson_project_corrections.docx)

I can order the parts and start making guide updates, but it would be good to have @jedgarpark do a review first. 

```
Walkmp3rson Project Corrections and Additions

1. Parts List Corrections and Additions
   - Adafruit I2C Stemma QT Rotary Encoder
   - Micro Potentiometer Knob (not needed)
   - Additional components needed: M2.5 Nylon hardware, heat shrink, Dupont connectors, CAT5 cable or 24AWG wire

2. CAD Files and 3D Printing Options
   - Libraries and community centers
   - Commercial services like ShapeWays

3. Display Connection
   - Header attachment to Feather Doubler
   - Preparing display wires
   - Correcting wire connection based on circuit diagram

4. Display and Header Assembly
   - Bending pins for display attachment
   - Testing display connection

5. Amplifier Preparation and Connection
   - Socket attachment to amplifier
   - Wire attachment difficulties
   - Hot glue to secure connections

6. Headphone Output
   - Heat shrink for plug
   - Connection to amplifier

7. On/Off Switch Installation
   - No specific on/off labeling
   - Hot glue for security

8. NeoKey and Encoder Testing
   - Cable connections

9. Final Assembly Adjustments
   - Display mount adjustments
   - Feather Doubler and USB port alignment
   - Button and switch mounting
   - SD card extender stability
   - Final assembly and screw placement

10. Code Corrections and Enhancements
    - Memory error on track switching
    - Directory structure for music organization
```

jedgarpark, 2024-03-01:
thanks @mikeysklar I'll have a look

jedgarpark, 2024-03-01:
@mikeysklar this will take some time to go through, but here's the first section I tackled:

> Walkmp3rson Project Corrections and Additions
> 
> 1. Parts List Corrections and Additions
>    - Adafruit I2C Stemma QT Rotary Encoder

This is already in the parts list, i bumped the quantity up to two

>    - Micro Potentiometer Knob (not needed)

I marked this as optional

>    - Additional components needed: M2.5 Nylon hardware, heat shrink, Dupont connectors, CAT5 cable or 24AWG wire

I added the dupont connectors we carry to the parts list, as well as the m2.5 hardware kit, heat shrink, and hook up wire, so ignore all of those

> 
> 2. CAD Files and 3D Printing Options
>    - Libraries and community centers
>    - Commercial services like ShapeWays
> 
I added an alert box about 3D printing services

> 3. Display Connection
>    - Header attachment to Feather Doubler
>    - Preparing display wires
>    - Correcting wire connection based on circuit diagram
> 
Not sure what didn't work for them based on the photos in the guide, mine is soldered that way and works, but a great point to follow the circuit diagram. I added another note to tell people to trust the circuit diagram above photos

mikeysklar, 2024-03-02:
Thank you for the updates. I think your commentary on the changes you have made is helpful for the submitter and myself.

jedgarpark, 2024-03-02:
>  3. Display Connection
> 
>    - Header attachment to Feather Doubler
Added a note to be sure to use the short headers
>    - Preparing display wires
>    - Correcting wire connection based on circuit diagram
photos in guide are of working board, so not sure what was causing issues, but I added a note to definitely follow the circuit diagram when in doubt
> 
> 4. Display and Header Assembly
>    - Bending pins for display attachment

i wouldn't bend them, use the right angled headers instead as mentioned in the guide. i added these to the parts list

>    - Testing display connection
> 
> 5. Amplifier Preparation and Connection
>    - Socket attachment to amplifier

i used right angle header  in the guide was to keep it low profile

>    - Wire attachment difficulties
>    - Hot glue to secure connections

good idea

> 
> 6. Headphone Output
>    - Heat shrink for plug
>    - Connection to amplifier
> 
> 7. On/Off Switch Installation
>    - No specific on/off labeling
>    - Hot glue for security
> 
> 8. NeoKey and Encoder Testing
>    - Cable connections
>

ashleymcnamara, 2025-03-29:
@mikeysklar you mention in your doc that the code crashes with a memory error, do you have the code changes?

mikeysklar, 2025-03-29:
@ashleymcnamara - this was from last year where some university students had written up their experience with the walkmp3rson. jdgarpark made some changes to the guide at the time. 

If you are encountering an issue please open a thread [on the forum](https://forums.adafruit.com/).