adafruit/Adafruit_Learning_System_Guides #2801: Error in IR example for CPI

bradanlane, 2024-05-02:
In the learning guide for the Circuit Playground Express, IR CPX to CPX example, the code has the following:

```
encoder = adafruit_irremote.GenericTransmit(header=[9500, 4500], one=[550, 550], zero=[550, 1700], trail=0)
```

However, the sample code from `adafruilt_irremote.py` has the following:

```
encoder = adafruit_irremote.GenericTransmit(header=[9000, 4500], one=[560, 1700], zero=[560, 560], trail=560)
```

The CPX example results in the receive data being the 1s compliment of the send data.

dhalbert, 2024-05-04:
Could you submit a PR for this? Thanks. That would be the easiest way to make sure it's corrected properly.

bradanlane, 2024-05-05:
OK, I forked the learning guides and went to make the change and found the same error in four different guides.

This has caused me to be concerned. I can only assume one of three possible situations:
1. those guides were not tested _(seems unlikely)_
2. those guides somehow work but only with themselves _(i.e. they somehow have the reciprocal error of the receive ... but I could not find that to be the case)_
3. something has changed since those guides were written _(most are 2+ years old)_

I am happy to submit the pull request but I would like another set of eyes to scrutinize the changed code. I do not have a Circuit Playground Express on hand and do not have the irRobotControl board(s).

I performed my testing using a Raspberry Pi PICO and the Adafruit IR Receiver board and IR Emitter board.

closed 2026-11-07 by bradanlane