adafruit/Adafruit_Learning_System_Guides #3120: Fruit_Jam_Product_Overlay TypeError without Wifi Configuration

relic-se, 2025-09-17:
If wifi credentials have not been configured within your device's environment variables, `fruitjam.fetch()` can also through a `TypeError` exception with the following details:

```
Traceback (most recent call last):
  File "<stdin>", line 158, in <module>
  File "adafruit_fruitjam/__init__.py", line 302, in fetch
  File "adafruit_portalbase/network.py", line 552, in fetch
  File "adafruit_portalbase/network.py", line 406, in connect
  File "adafruit_portalbase/wifi_coprocessor.py", line 85, in connect
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 657, in connect
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 680, in connect_AP
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 548, in wifi_set_network
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 421, in _send_command_get_response
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 303, in _send_command
TypeError: object of type 'NoneType' has no len()
```

`TypeError` could be added to the tuple on this line:
https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/b373170c416dc3222b03228c042c9bbad14f66d1/Fruit_Jam/Fruit_Jam_Product_Overlay/code.py#L169

Or something along the lines of the following exception handler could be added to inform the user:
``` python
except TypeError as e:
    print("WiFi has not be configured properly.")
    supervisor.reload()
```

FoamyGuy, 2025-09-17:
I would be in favor of changing this at the portalbase layer to raise an exception that isn't a `TypeError`. It makes literal sense because the code at the bottom of the trace is calling `len(None)`, but it's not really a useful description of the true cause of the error, missing WIFI credentials in settings.toml. 

CPython requests will raise `requests.exception.ConnectionError` ultimately if an attempt to make a request is made when there is no active network connection
```
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='httpbin.org', port=443): Max retries exceeded with url: /get (Caused by NameResolutionError("<urllib3.connection.HTTPSConnection object at 0x7086577971a0>: Failed to resolve 'httpbin.org' ([Errno -3] Temporary failure in name resolution)"))
```
So I'd lean towards `ConnectionError` as the type for this error. It can also contain a message letting the user know to fill in WIFI credentials to settings.toml.

Fixing in portalbase instead of this project specifically means that other projects based on that library get the same improved error behavior.

relic-se, 2025-09-17:
@FoamyGuy Here's my approach to the fix within `adafruit_portalbase`: https://github.com/adafruit/Adafruit_CircuitPython_PortalBase/pull/116

FoamyGuy, 2025-10-03:
This is resolved now by https://github.com/adafruit/Adafruit_CircuitPython_PortalBase/pull/116 and merged.